OpenIPSL/OpenIPSL #270: Which parameters should be activated for GENCLS and which are initial

dietmarw, 2021-04-23:
The model `OpenIPSL.Electrical.Machines.PSSE.GENCLS` has currently all power flow variables deactivated which is probably not quite correct. This is also demonstrated in https://youtu.be/DZ_hcUXPjMA?t=745 

The core reason of the deactivation of the power flow parameters stems from the misleading/missing documentation of the component itself. It has the name "inf" on the icon  which suggests this is simply an infinite bus. But actually it is just a simplified synchronous generator that **can** function as an infinite bus **if** certain parameters are set correctly. 

With that in mind it makes  now perfect sense to have also other power flow parameters activated than just `v_0` and `angle_0`. 

To fix this I suggest the following changes to `GENCLS`:
 - remove the misleading "Inf" string from the icon and place a `%name` label instead
 - activate all power flow variables except for `V_b` and `f_n` (which still have no function in that model)
 - add a description string and also add documentation how this classic model can be "misused" as infinite bus. 
 
@"sergio", @marcelofcastro   Would it not be better to use a proper infinite bus like (`OpenIPSL.Electrical.Buses.InfiniteBus`) for the SMIB example?

marcelofcastro, 2021-04-23:
GENCL, as you said, is a model that has two functions: it should be capable of representing a classical machine model when H is set >0 and it should represent an infinite bus if H = 0. This is exactly how it is done in PSSE and this is why we need to represent infinite buses using GENCLS if we want to compare results with PSSE (so that would be the reason why the answer to your last question is no).

So my suggestion is:

> remove the misleading "Inf" string from the icon and place a `%name` label instead
- We could allow `%name` if `H>0` and `Inf` if `H = 0` (on a practical implementation that would mean < eps). Is that possible @dietmarw ? We could even change the icon to something that could represent both.

> activate all power flow variables except for `V_b` and `f_n` (which still have no function in that model)
- I think it might be better if we allow power flow variables like P and Q to be accessed only if `H>0` and allow power flow variables like V and Angle at all times.

> add a description string and also add documentation how this classic model can be "misused" as infinite bus.
- We should improve the documentation for sure, because although we have one sentence saying "Synchronous machine represented by 'classical' modeling or Thevenin voltage source (for H = 0)", this is not enough for understanding the model's purpose and correct usage as a classical machine or as an infinite bus.

dietmarw, 2021-04-23:
@marcelofcastro Your suggestions sound good although I would like to always have `%name` visible on the icon (on all models in OpenIPSL actually) so that one can see the instance names right away. Helps with plotting etc and some models where the instance names are unfortunate. The conditional display is possible though will not be supported in OMEdit (yet one hopes). 

I'll start a PR and post a link here soon.

As for the usage inside @sergio-dorado 's tutorial it still makes me question why you would need/want to initialise an inf bus with P and Q as that would be commanded by whatever is connected *before* the inf bus.

lvanfretti, 2021-05-21:
@dietmarw some feedback here. In all power system programs, the infinite bus only applies for power flow. What the programs do internally is to replace the infinite bus with a GENCLS with a very high inertia to "fix" the frequency of the system. Next, another use of the GENCLS is to approximate external areas, in that sense you will set the fixed P and Q to a specific value being either absorved or delivered, for example to model the power going from the "local" detailed model to an "external" area.
This sometimes is also done with a load, where you would specify the P and Q so that it delivers a certain amount of power.
There are some examples where we have done that to remove equipment, for example and HVDC line, where you will replace the sending end with a load and the receiving end with a generator, both of which have the set powers from the line.
So, what @sergio-dorado is doing in the tutorial is exactly specify the P and Q that should be injected or absorved by the local system into the network, depending on the power flow condition of the external system.

These are very common things that are done by people that use PSS/E, PSAT or PST, to name a few.

Finally, GENCLS is also used sometimes when there is very limited information on how to model a specific machine, but not to represent an external system.

Of course, in Modelica, we can force the frequency to be set to a fixed value with the infinite bus, but that doesn't allow to represent when a specific P and Q should be either absorved or delivered.

I would prefer that we leave the GENCLS internal modeling capabilities to be able to represent the "large" generator and/or a specific machine.

So, the available options are good to have in general. What we need is to document this type of usage in the corresponding classes. I can provide some literature about this if you think this is a good idea, while it is common knowledge with power system specialists, it seems like it is not obvious.

I can explain more next time we talk, next week.

dietmarw, 2021-05-25:
OK I understand. So I propose to change it as such that it does not claim to be a infinite bus but document the two modes of operation:
 1. Infinite bus with `H=0`, which means that `v` and `f` can be parametrised and stay fixed. Parameters `P` and `Q` have no meaning. 
 2. Large Generator (`H>0`), which means that `P` and `Q` in addition to now only **initial** `v` and `f` can be parametrised. `P` and `Q` would then stay fixed just like for an ideal load/generator unless of course `v` changes.
Is that analysis correct?

lvanfretti, 2021-05-26:
The analysis is correct, I would propose the following options and wording:

1. Instead of saying that P and Q have no meaning, maybe we can state: `P` and `Q` will be computed from the model to make `v` and `f` constant.

2. ``Classical Generator'' model with physically-meaningful parameters, such as `H>0` (e.g. `H = 5` in a 100 MVA base) ... continue the description above ...  Application: Can be used to model a generation facility when only limited information is available, this is the simplest model possible to represent a synchronous machine's electro-mechanical dynamics.

3. External grid with `H>>0' (e.g. `H = 999999.99` in a 100 MVA base) ... continue the discussion above. 

Application: Can be used to model the connection to a ``stiff grid''. By making `H` very large, the speed of the machine will not undergo any changes, and consequently will set the frequency `f` at the bus to a fixed value (e.g. 50/60 Hz). Moreover, this allows to specify a value of `P` and `Q` that is either injected or absorbed.

dietmarw, 2021-05-26:
I've created #273 which should fix this and also make GENCLS parametrisable again. @sergio-dorado please have a look too.

PR #273 (2021-06-03): Improves documentation of GENCLS and unlocks parameters
Fixes #270 by improving the documentation and also opening the parameters that were locked before.
closed 2021-06-03 by dietmarw