OpenIPSL/OpenIPSL #277: GGOV1 variant for linearization or update using Pade model

lvanfretti, 2021-07-31:
@ManuelNvro and @dietmarw 

@sergio-dorado reported an issue over email that I will add below.

When attempting to linearize certain blocks (e.g. the delay=`Modelica.Blocks.Nonlinear.FixedDelay` model), any Modelica tool should fail.  This is expected when using that model as it is not possible to linearize it, it does not use either functions for derivatives or a smooth operator-based implementation.

Most user's of this library will not understand this, and think there is a problem with the tool or the library.

We have two choices, 

1. either create model variants that are "fit for linearization" that can be easily replaced, 
2. or that we replace all "delay" blocks with a Pade approximation from (`Modelica.Blocks.Nonlinar.PadeDelay`) 

@marcelofcastro if we replace with the Pade approximation we have to run all the CI tests to made sure we have the new results for all models that have the change...

My preference would be for no. 2, because I don't think the user's will pay attention to the availability of a variant, but that of course gives more work in running CI tests.
What do you think @marcelofcastro and @dietmarw?

Issue report from @sergio-dorado (please do this using the Issues tab next time!):

> ---------- Forwarded message ----------
> From: Sergio A. Dorado-Rojas <sergio.dorado.rojas@gmail.com>
> Date: Jul 31, 2021, 7:53 AM -0400
> To: Marcelo de Castro Fernandes <decasm3@rpi.edu>, Giuseppe Laera <laerag@rpi.edu>
> Cc: Luigi Vanfretti <luigi.vanfretti@gmail.com>
> Subject: Issues with Delay Block for Linearization
> 
> Hi all,
> 
> I have been working with an OpenIPSL model for my internship. I was trying to perform a linear analysis to change some AVR’s parameters and found something that I think is worth to discuss.  The model has a single generator with TG (GGOV1) and ES (ESST4B). Marcelo did the translation for me so it is working on the latest OpenIPSL version. I did not create this as an issue on GitHub because ... **redacted.**

[image]
> 
> Whenever I try to use any function within the Linear Analysis menu, I get this message. The linear analysis is not carried out.
> 
[image]
> 
> 
> So, I traced down the model for delay blocks and found that there is one in the turbine model. 
> 
[image]

> 
> 
> For my particular case, the delay time was exactly zero so I removed the block. The linear analysis functions could be used perfectly afterwards.
> 
> In my humble opinion, it doesn’t speak highly for the library that the linear functions cannot be used for such blocks. I’d suggest to replace the linear block by a Padé approximation. However, I don’t know much of how this would impact the exactitude of the model against its PSS/E counterpart. I haven’t tried this workaround myself since I am not familiar at all with this particular exciter’s structure and, as I mentioned above, for my immediate needs it was better to remove the block. 
> 
> Let me know what your thoughts are. 
> 
> Best,
> Sergio

marcelofcastro, 2021-08-01:
couldn't we just set the environment variable, as suggested by the log? I guess that replacing the delay block, which is supposed to be the correct way of doing it, by a Pade approximation, for example, would be just constraining a model of its full behavior just for the sake of linearization. It doesn't seem wise to me to fully modify the block. In addition, if we can just set the environment variable for the linearization and continue with the block the way it is right now might be the easiest solution. That said, if we really want to address the issue by changing the turbine representation, I guess we can use a block that can be replaced by Delay or PadeApprox depending on a "choice" parameter, where the user can select what they want: the correct implementation or the Pade block. @dietmarw what's on your mind?

lvanfretti, 2021-08-01:
I agree with using the choice and allowing to select between models, that seems to be the best way forward.

Luigi

dietmarw, 2021-08-09:
I agree with @marcelofcastro that we really should not restrict the dynamic models for the sake of linearisation which kind of defeats the powers of the dynamic simulations possible in Modelica. So if one wants to linearise it for a certain application then this should happen by using the suggested env variable offered by the tool and not changing the model implementation itself.

lvanfretti, 2021-08-16:
Marcelo and Dietmar

I did a bit more digging, I created a simple model to reproduce the error.
Dymola’s developer has confirmed the way of setting the flag.

Now, you can see from his response below that when using the flag the delay is approximated assuming a very large time horizon and suggests that we use a Pade approximation.

So, I think we should have a “choice” for the models at the top level where we have the delay part as a replaceable model and it can use the Pade for cases like this.

Best

Luigi

dietmarw, 2021-09-03:
I've checked and the `FixedDelay` is used only in:
- `OpenIPSL.Electrical.Controls.PSSE.TG.BaseClasses.GGOV1.Turbine`
- `OpenIPSL.Electrical.Solar.KTH.PFblocks.MPPT`
The latter is probably not even present any more after merge of #267 
So I think it is overkill to add a global parameter just to trigger **one** change in **one** model. Instead what I suggest we can do is to make the `delay` block replaceable and constrain the replacement choice to `PadeDelay`. That way one can make the change in the model for linearisation if it is needed.

lvanfretti, 2021-09-03:
Sounds like a great approach.
We should also add default parameters for the delay, let me look that up when I get to the office.

dietmarw, 2022-05-11:
@lvanfretti I've just created a fix for this and all that is missing now is some nice defaults which we can add as documentation in the info layer. Did you manage to get to your office in the mean time ;-)

lvanfretti, 2022-05-11:
@dietmarw working from home today buddy! I'll go an approve that fix, thank you!

closed 2022-05-12 by lvanfretti

dietmarw, 2022-05-12:
Yes, make a new ticket for 3.1.0 (for now). The selection of models can be done via parameter (already done in `OpenIPSL.Electrical.Machines.PSSE.Plant`) but question is how often will this be needed? Anyway, have a new ticket and continue the discussion there.

dietmarw, 2022-05-12:
Actually! It is a VERY simple fix by adding "choicesAllMatching=true"

lvanfretti, 2022-05-12:
I tried that but was not behaving how I was expecting it to if you see the removed commented text in [fea08e4](https://github.com/OpenIPSL/OpenIPSL/pull/288/commits/fea08e4bca3d71822ff956933044d33889dae9d1), maybe you have better luck? I'll open the other issue anyway.

PR #288 (2022-05-12): Add option to use PadeDelay instead of normal delay to be used for linearisation
Fixes #277
commit 168d3a75 (2022-05-12): Make delay block replacable to allow use of PadeDelay

Fixes #277