OpenIPSL/OpenIPSL #282: Enhancements from meeting with user

lvanfretti, 2021-12-09:
Hi @dietmarw and @marcelofcastro 

I had a meeting with one of our users, Prof. Massimo Ceraolo, @max-privato , he provided me the following valuable feedback.
These are my notes, @max-privato, could clarify further if needed.

@marcelofcastro do you think we can apply these for a future release? In such case, which one?

General Enhancements:
- Icon view for base connector should be consistent, p and n pins should have different appearance.
- Be consistent with the MSL electrical connector convention.
- Luigi’s note: Change tone of color to make a visual difference, similar to what DS does with the EPS library, darker colors.

Specific:

Generators:
- Improve display: `OpenIPSL.Electrical.Machines.PSSE.GENSAL`
- Increase text box width of output and input names to improve rendering.
- Text should be right justified for names on the right and vice versa for those on the left.

Lines:
- Improve convention on line model: `OpenIPSL.Electrical.Branches.PwLine`
- Change icon to define what is terminal 1 and terminal 2 as used in the line flows. This helps relate what is P12 and Q21 computed inside of the line model from the schematic view.
- Line parameters: How is the user supposed to know the voltage base for the per unit parameters of the lines?
Add explanation on the User’s Guide how it is done using one of the existing examples, (e.g. unit test).

Transformer:
- Improve convention on transformer model: `OpenIPSL.Electrical.Branches.PSSE.TwoWindingTransformer`
- In the text layer, we define terminal p and n, however, the input data refers to terminals 1 and 2, this is inconsistent and hard for the user to figure out which side is which. 
- Suggestion: Change icon to identify on diagram layer easily to identify what is 1 and 2.

max-privato, 2021-12-19:
I would like to express these recommendations in a way that shows that there a common ground around them.

**1) positive and negative pins.**

Positive and negative pins should have different appearances, exactly as it occurs in all the two-terminal components in MSL. this would help to interpret output quantities, i.e. voltages and currents (both terminals have a voltage and current of their own, differing from each other on their sign)
Compare the two diagrams below. Using the diagram below when the user looks for instance at the data of pwLine, it will immediately understand pwLine.p .i and pwLine.n.i to which terminals belong.

[image]

[image]

**2) Transformer windings names**
In the parameter dialog box we see winding 1 and winding 2. But on the diagram, we see pin "p" ad pin "n". So, the user does not know which is which. I propose that the dialog box should mention 'winding 1 ("p")' and 'winding 2 ("n")'. It will keep names 1 and 2 for psse compatibility but will be also meaningful for any other user

**3) generator icons**
I just found that nearly always the names of the generator ports are shown with ellipses, just because the text boxes are too small and badly adjusted. Compare the two appearances below, built using the present text boxes (upper picture) and enlarged and correctly justified ones as I propose (lower figure).

[image]

[image]

 Making this change implies just a few minutes. 
Below the modified  BaseClasses.baseMachine icon, for your reference:

`   Icon(coordinateSystem(preserveAspectRatio=false, extent={{-100,-100},{100,
            100}}), graphics={Rectangle(extent={{-100,100},{104,-100}},
          lineColor={0,0,255},
          fillColor={255,255,255},
          fillPattern=FillPattern.Solid),
                               Text(
          extent={{10,76},{98,65}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="SPEED",
          horizontalAlignment=TextAlignment.Right),
                              Text(
          extent={{10,-66},{98,-76}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="ISORCE",
          horizontalAlignment=TextAlignment.Right),
                               Text(
          extent={{6,-24},{98,-35}},
          lineColor={0,0,255},
          textString="ETERM",
          horizontalAlignment=TextAlignment.Right),
                              Text(
          extent={{8,94},{98,84}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="ANGLE",
          horizontalAlignment=TextAlignment.Right),
                              Text(
          extent={{8,56},{98,46}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="PMECH0",
          horizontalAlignment=TextAlignment.Right),
                               Text(
          extent={{6,34},{98,24}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="PELEC",
          horizontalAlignment=TextAlignment.Right),
                              Text(
          extent={{8,-46},{98,-56}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="EFD0",
          horizontalAlignment=TextAlignment.Right),
                             Text(
          extent={{6,-86},{98,-96}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="XADIFD0",
          horizontalAlignment=TextAlignment.Right),
                                Text(
          extent={{-92,60},{-6,42}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="PMECH",
          horizontalAlignment=TextAlignment.Left),
                               Text(
          extent={{-92,-40},{-8,-60}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="EFD",
          horizontalAlignment=TextAlignment.Left),
                                Text(
          extent={{-100,130},{100,108}},
          lineColor={0,0,255},
          lineThickness=0.5,
          textString="%name")}));
`

**4) the pu system.**
Consider the example Examples.Branches.PSSE.TwoWIndingTransformer. the user has lines whose parameters are in p.u. What is the p.u. base? The power is obvious the System base power (100 MVA in this example) but what is the voltage? After some thought I figured out that OpenIPSL internally works with pu quantities, and therefore in this example, the nominal voltage of the lines is the one of the secondary of the transformer. Nevertheless, the user does not know whether he must refer to the _nominal_ or _base_ voltage (they could be different, as in the primary of the transformer of this example). So, without some explanation in the documentation, it is very difficult to figure how to use the pu system. This could be even more tricky in case of meshed lines, since a line can have at its two ends transformers whose nominal voltages on their winding in connection with that line, could be different from each other.

----------------
Finally, not mentioned by @lvanfretti , I think that the name of the lines when the power flow data is not visible should appear outside the line rectangle, and be larger than they are now: they're too small. making text to toggle between "show power flow data - hide line name" and "show line name hide power flow data" should be straightforward in Modelica.

lvanfretti, 2023-01-23:
Moving materials from #355.

I have received an email from @max-privato with the folowing suggestions for graphical enhancements, I'm opening this issue so the suggestions and be discussed.

From email, ad verbum:

1. better readability of the line names (this is obtained eliminating the grey boxes inside)
2. better display of texts (check for instance generator names and load powers)
3. visual differentiation of positive and negative pins of components (as in MSL), so that the output variables are more effectively associated to points of the diagram.

Screenshot from 3.0.1:
[image]

Screenshot after modifications:
[image]

lvanfretti, 2023-01-23:
Clarification from @max-privato :
> 
> This IS NOT a duplicate of 335: it contains new ideas: change of the line
> appearence, and of the load data.
> *More important*, this contains a screenshot of a prototype implementation
> showing a possible outcome, which could be totally lost if this ticket is
> disregarded.

lvanfretti, 2023-01-23:
@max-privato and @dietmarw I changed the topic of this PR as Graphical Enhancements and now we can document both previous and new ideas on this ticket. Nothing will get lost as long as we have some clarification.