OpenIPSL/OpenIPSL #289: GGOV1 - Enhancements - Enable variants for linearization from the top layer

lvanfretti, 2022-05-12:
Follow-up from #277, this issue is to document a potential enhancement, for user friendliness to implement in future releases.

- [ ] Provide means to select which model to replace by, i.e. select between Pade or Fixe, from upper level (GGOV1) layer, through some means. Via parameter (?).
- [ ]  Create test model for Pade variant.
- [ ] Create application case. This should be an example to show how to compare model variants in time-simulation, compare linearized vs non-linear variant model responses.  

The last item is more as an excuse of showing potential problems of linearization when sub-components are non-linearizable, hence, improving this model can help provide a guideline on creating model variants for any other sub-components needed in models that require a differentiable variant.

lvanfretti, 2022-05-12:
See: 
- https://github.com/OpenIPSL/OpenIPSL/issues/277#issuecomment-1124972847 and
- https://github.com/OpenIPSL/OpenIPSL/issues/277#issuecomment-1124975699
For future discussion.

dietmarw, 2022-05-12:
I've started #290 which now adds (in Dymola) the following dialogue: 
[image]

dietmarw, 2022-05-12:
I forgot to propagate the choice also upwards into the `GGOV1` and `GGOV1DU` models. That's done now.

lvanfretti, 2022-05-12:
@dietmarw see testing issue on https://github.com/OpenIPSL/OpenIPSL/pull/290#issuecomment-1125071324

dietmarw, 2022-05-24:
I think we should create a new issue for 
"
 Create application case. This should be an example to show how to compare model variants in time-simulation, compare linearized vs non-linear variant model responses.
"
What do you think @marcelofcastro

marcelofcastro, 2022-05-24:
I think it's a good idea to do that, @dietmarw. So we can set the 3.1.0 milestone to that issue, right?

dietmarw, 2022-05-24:
Yes, exactly. And close this issue here.

marcelofcastro, 2022-05-24:
Sounds good!

marcelofcastro, 2022-05-27:
The last task here is now tracked in issue #292 and I am closing this one.

closed 2022-05-27 by marcelofcastro