OpenIPSL/OpenIPSL #296: PSSE Model Tracker

marcelofcastro, 2022-06-06:
This issue is created to track the addition of PSSE models into OpenIPSL. They are:

Machines:
- [ ] GENTPJ

Exciters:
- [ ] AC7B
- [ ] AC8B
- [ ] DC4B
- [ ] ESURRY

Excitation Limiters/ Compensators:
- [ ] IEEEVC
- [ ] MNLEX2

Turbine Governors:
- [ ] IEEEG2
- [ ] WEHGOV
- [ ] WPIDHY
- [ ] WSIEG

Renewable Sources:
- [ ] REGCA1
- [ ] REECB1
- [ ] REECCU1
- [ ] REPCA1

dietmarw, 2022-06-08:
Some of the models are already available as drafts in https://github.com/marcelofcastro/OpenIPSL/tree/add-new-models

marcelofcastro, 2022-06-08:
True @dietmarw, but I am so confused by that branch that has been long forgotten that I'd prefer to prepare another branch with those models for a PR. Or maybe working with that branch but with way more caution.

lvanfretti, 2024-08-16:
Added links to PRs and repos where the new components are.

dietmarw, 2024-11-14:
I've updated the list of finished PSSE models after #347 was merged. @marcelofcastro could you create a PR based on your marcelofcastro/add-new-models branch so that we can continue improving those and also track them?

gkhalley, 2024-12-02:
We use GENTPJ1 instead of GENTPJ (not an option). These may be the same in OpenIPSL, but I would prefer that it be labeled with the additional 1 if it is.
[image]

lvanfretti, 2026-10-31:
**TL;DR.** This tracker is replaced by #436, which carries over the whole checklist and every comment posted here, and orders the PSS/E work in four topics (debug, integrate, validate, develop) with the models users ask for at the top. Closing this one so there is a single list to maintain.

**Key points**

- Everything from this issue is in #436 under "Carried over from #296": GENDCO, EXAC3, IEEET3, REXSYS, ST6B, ST7C (STC7 here) still open; the rest merged through #347 and #327.
- The GENTPJ1 naming request of 2024-12-02 is answered there (A.1, A.7): the class keeps its name, the description string and documentation state the PSS/E model name.
- New requests from the same user (REGCA1, REECC1, REPCA1 validation; VTGTPAT, FRQTPAT, CMLDBLU2, CMLDZNU2, IEELOW) are in #436 too.

Generated with Claude Code

closed 2026-10-31 by lvanfretti