OpenIPSL/OpenIPSL #305: Annotation for DAE Mode in OM's Coverage Testing

lvanfretti, 2022-07-04:
According to recent correspondence with the OM team, we can add the following annotation to Examples and Test models if we want to have them simulated with daemode:
`annotation(__OpenModelica_commandLineOptions = "--daeMode");`
Results from coverage in OpenModelica are listed here:
https://github.com/OpenModelica/OpenModelica/wiki/Library-Coverage-Testing
Master Branch:
https://libraries.openmodelica.org/branches/master/OpenIPSL_dev/
Latest Release:
https://libraries.openmodelica.org/branches/master/OpenIPSL_release/
@marcelofcastro can you add the annotations for the next release?

dietmarw, 2024-11-16:
@marcelofcastro  I've seen this annotation in various places now. Are they present where they need to be or are some still missing?

lvanfretti, 2024-11-17:
@dietmarw I think we met in 2023 with other people using OpenIPSL, and we promised them that we would add this flag where it would be needed.

They should mainly be [here](https://github.com/OpenIPSL/OpenIPSL/tree/master/OpenIPSL/Examples/DAEMode) , as that has the key examples that benefit from it, but perhaps we also need to put it in [here](https://github.com/OpenIPSL/OpenIPSL/tree/master/OpenIPSL/Examples/N44), as it pertains the original version of the Nordic44 model.

I am not sure where else they are, but these are the main ones where it should be.

I am wondering that since we are setting this as default for OM, shouldn't we also make it default for Dymola? I am not entirely sure we can put the ones of Dymola as annotations, as I usually enable those as flags within a function such as the one: [here](https://github.com/ALSETLab/CCTA-OpenIPSL/blob/main/Example2/Utilities/SetupSolverSettings/On.mo)

dietmarw, 2024-11-18:
> I am wondering that since we are setting this as default for OM, shouldn't we also make it default for Dymola? I am not entirely sure we can put the ones of Dymola as annotations, as I usually [...]

I'm also not sure if `Advanced.Define.DAEsolver = true` can be passed by Dymolas vendor annotation. @HansOlsson is that possible?