OpenIPSL/OpenIPSL #317: Connect on non-connectors

maltelenz, 2022-10-31:
In `OpenIPSL.Electrical.Solar.PowerFactory.General.ElmGenstat` there is:

```
    connect(sinref, sinu);
    connect(cosref, cosu);
```

But neither `sinu` nor `cosu` are connectors.

System Modeler reports this when validating `OpenIPSL.Tests.Solar.PowerFactory.DIgSILENT_PV`:

> Error: OpenIPSL.Electrical.Solar.PowerFactory.General.ElmGenstat [21:5-21:5] Invalid connect equation. cosu is not a connector.
>    Component references in connect equations must be of the form c_1.c_2...c_n or m.c_0 where c_i is an instance of a connector class for all values of i.

> Error: OpenIPSL.Electrical.Solar.PowerFactory.General.ElmGenstat [20:5-20:5] Invalid connect equation. sinu is not a connector.
>    Component references in connect equations must be of the form c_1.c_2...c_n or m.c_0 where c_i is an instance of a connector class for all values of i.

lvanfretti, 2022-11-02:
Thanks @maltelenz.

I guess I'll have to take a look a this one. This is a very old model and corresponds to an implementation done in a MSc thesis of a generic model before the reference tool had a native PV model. Now we have a pull request #267 that is an reimplementation of a native model in Modelica, so I am wondering if we should simple get rid of the "user defined" model and keep only the one in #267. What do you think @dietmarw ?

dietmarw, 2022-11-04:
I like to forward the question to @tinrabuzin. Do you think the `ElmGenstat` model can/should be removed in favour of the models that you created as part of #266?

tinrabuzin, 2022-11-04:
@dietmarw 

I think that `ElmGenstat` was actually cleaned-up by me in #266 and I vaguely remember why I had this connect on non-connectors. I think I wanted to be able to switch between PLL-computed angles and the actual angles.

I don't know if we can just write the following instead?
```
    sinref = sinu;
    cosref = cosu;
```

tinrabuzin, 2022-11-04:
Also, just FYI, the models in #266 are not the same as in #267, i.e. we cannot just replace the ones from #266 with the one in #267. The former is a much more detailed mode while the latter is a model of distributed PVs.

dietmarw, 2022-11-04:
So the way I understood it, so are #267 additional models.

dietmarw, 2022-11-04:
> I don't know if we can just write the following instead?
> 
> ```
>     sinref = sinu;
>     cosref = cosu;
> ```
No, the solution is to make sinu and cosu protected outputs. However, there seems to be an even more sever problem. Running `Tests.Solar.PowerFactory.DIgSILENT_PV` gives: 

Model error - division by zero: (1.0) / (pv_plant.busbar.integrator.y) = (1) / (0)

Which is not surprising given that `pv_plant.busbar.integrator.y` is initialised as 0. Question is, what should be the initial value be?

lvanfretti, 2022-11-04:
@dietmarw , #266 was originally implemented by Farhan in PowerFactory when it did not have any PV models and then I got him to work in iTesla to implement it in Modelica for the library. The original PowerFactory implementation is documented here: http://urn.kb.se/resolve?urn=urn:nbn:se:kth:diva-105746

So, I believe this model should be under .Solar.PowerFactory.Generic instead of under .Solar.PowerFactory.DigSILENT since it is not corresponding to the "native" implementation in that reference tool.

Meanwhile #267 seems to be a Modelica implementation of the native model in PowerFactory. Right @tinrabuzin. So, that should be the only one under .Solar.PowerFactory.DigSILENT, until new ones are added.

tinrabuzin, 2022-11-04:
The old model was unusable. The model in #266 is according to the existing native implementation from DigSILENT.

dietmarw, 2023-01-12:
@maltelenz Have a look at #332. That should fix it.

dietmarw, 2023-01-12:
@marcelofcastro Fixed. Have another look.

closed 2023-01-13 by marcelofcastro

maltelenz, 2023-01-17:
> @maltelenz Have a look at #332. That should fix it.

It is indeed fixed and the error gone when validating in System Modeler. Thanks!

PR #332 (2023-01-13): Convert internal variable to connector
Fixes #317 by using protected connectors instead of helping variables.
commit 3b280ad2 (2023-01-13): Convert internal variables to connectors

This allows conditional connection and fixes #317