OpenIPSL/OpenIPSL #334: MVA as displayUnit for V.A

maltelenz, 2023-01-17:
Here:
https://github.com/OpenIPSL/OpenIPSL/blob/9879395c5a1ae19619f20ac8bced2bceedf999a0/OpenIPSL/Types.mo#L9

a displayUnit of `MVA` is defined for `Modelica.SI.ApparentPower`, which has:

```
type ApparentPower = Real(final quantity = "Power", final unit = "V.A");
```

Should it be `displayUnit = MV.A` to be compatible?

dietmarw, 2023-01-18:
I would argue that the **display**Unit is meant for display. Hence having `MV.A` would make it rather strange when it is almost exclusively handled as MVA.  Just the same way as the reactive power is `var` and not `V.Ar` or should it then be `V.A.r`?

maltelenz, 2023-01-18:
Without having any experience with these units (this is not my field), there is a difference.

`var` is the unit in the standard library, so it goes nicely with `Mvar` as a display unit, only adding a prefix.

For `VA`, the unit in the standard library is `V.A` (after https://github.com/modelica/ModelicaStandardLibrary/pull/2982/files ), so the "natural" display unit that adds a prefix would be `MV.A`. Otherwise how is a tool expected to know how `V.A` and `MVA` are connected, and convert accordingly?

From my admittedly limited understanding, you also wouldn't do this, which seems similar:

```
Real(unit = "N.m", displayUnit = "MNm")
```

dietmarw, 2023-01-23:
Mmh it looks like it really is something that the tools should be able to figure out. I just tested it in Dymola and there it is no issue that the base unit is defined as `MVA`. You get the choice of `V.A`, `MVA` and `MV.A`. So you can choose to your taste and it still manages to do the conversion for the display. 
So I'm still reluctant to specify `MV.A` when the normal *displayed* units are `MVA`.

henrikt-ma, 2023-01-23:
To me it looks more like a tooling thing to not show the Modelica dot in a displayed unit, if that's the preferred appearance.  As @maltelenz indicates, it seems like a real mess for tools to guess the meaning of display units if they don't parse according to what they mean.

Sure, a tool can provide a tool-specific non-SI symbol `"MVA"` that is defined as `"MV.A"`, but it becomes impossible for a tool to guess the complete set of such non-parsable symbols that might be encountered in libraries.  Also, having such tool-specific display unit symbols becomes a problem for portability across tools.

Let us please avoid the need for guesswork.

dietmarw, 2023-01-23:
You are right, since the tool can still decide to display `MVA` making it easier for others to derive the scaling would be better. This is now fixed.

maltelenz, 2023-02-16:
There are 23 more cases spread out through the library, see:
https://github.com/OpenIPSL/OpenIPSL/search?p=1&q=displayUnit+%3D+%22MVA%22&type=code

I don't seem to have the permissions to reopen the issue.

dietmarw, 2023-02-16:
Well spotted, reopening and those modifications should simply get removed since the type itself sets already the correct displayUnit

dietmarw, 2023-02-16:
@maltelenz Something is off, the search result that you have linked to is not coincide with a search on the files system.  There are only two places that are still using MVA in the repo (fixed now in : 55a1550af4b2508b3f6f965a58a49c7adcf2a144)

maltelenz, 2023-02-16:
Agreed, I have no idea how the search I linked works or where it found the extra cases. One of the places you fixed was the one I had actually discovered, so I think we are all good now. Thanks!

commit 99b18b15 (2023-01-23): Fixes #334 by using MV.A instead of MVA
closed 2023-02-16 by maltelenz