OpenIPSL/OpenIPSL #335: Graphical Enhancements

lvanfretti, 2023-01-23:
I have received an email from @max-privato with the folowing suggestions for graphical enhancements, I'm opening this issue so the suggestions and be discussed.

From email, ad verbum:

1. better readability of the line names (this is obtained eliminating the grey boxes inside)
2. better display of texts (check for instance generator names and load powers)
3: visual differentiation of positive and negative pins of components (as in MSL), so that the output variables are more effectively associated to points of the diagram.

Screenshot from 3.0.1:
[image]

Screenshot after:
[image]

dietmarw, 2023-01-23:
This looks like a duplicate of #282 right?

max-privato, 2023-01-23:
This IS NOT a duplicate of 335: it contains new ideas: change of the line
appearence, and of the load data.
*More important*, this contains a screenshot of a prototype implementation
showing a possible outcome, which could be totally lost if this ticket is
disregarded.

dietmarw, 2023-01-23:
@max-privato In that case, could you add those ideas to ticket #282. Since it is all related to the appearance it does not make sense to have it distributed over several tickets.

lvanfretti, 2023-01-23:
@dietmarw I'll do that since I'm the one who opened the new ticket.
It was not clear to me at first glance, but I now understand the difference on what @max-privato is proposing and I guess we'll get more clarifications and maybe even a PR from him if he already has the prototype implementaiton.