OpenIPSL/OpenIPSL #338: Tests without experiment annotations?

maltelenz, 2023-03-02:
I see there are models that look like they are intended to be proper tests in the `Tests` package, that don't have an experiment annotation.

Example:
https://github.com/OpenIPSL/OpenIPSL/blob/master/OpenIPSL/Tests/Branches/PSAT/TwoWindingTransformer_Test.mo

Are these models intended as full tests? Should experiment annotations with at least a stop time be added?

lvanfretti, 2023-03-09:
The `Tests` sub-package is intended for a kind of unit testing, extending from the partial model/template `OpenIPSL.Tests.BaseClasses.SMIB`, to check each component without worrying about "bugs" in the rest of the system.

However, some models do not inherit from this template that has pre-configured changes so that you can see dynamics, because we haven't included the variant for series and shunt components, only for plants the one above.
@marcelofcastro and @GiuseppeLaera could you please get this done? You developed the templates for the NYPA project, so they just need to be added to the library.

For the model in question, we need to modify it once we add the template models, which would have the same experiment annotation throughout unless necessary to change it for some particular component testing purpose.

In the mean time, you can do your testing with stop time of 1 sec or 10 sec, it really doesn't matter because that component (and the following two, i.e. ThreeW..., PhaseShift) do not have dynamics, so the simulation should only verify that they are flat (i.e. initialization works properly).

dietmarw, 2024-11-25:
Is anyone working on this. Just asking since we are kind of behind the planned release date already and either we decide to skip this for 3.1.0 or we need to fix this.

lvanfretti, 2024-11-26:
@dietmarw Nobody is working on it, I could do it after on the week of Dec 16th, after classes are done... I guess we would target before Christmas for the new release?

dietmarw, 2025-05-06:
@lvanfretti Christmas has long gone, so are the classes again ;-) so what is the status on this?

lvanfretti, 2025-05-06:
@dietmarw I completely forgot about this... I'll add this to the list from our recent meeting.