OpenIPSL/OpenIPSL #346: Problems solving initial system for Order5test2 in System Modeler

maltelenz, 2024-02-08:
System Modeler has trouble solving the initial system for `OpenIPSL.Tests.Machines.PSAT.Order5test2`:

```
The initialization problem is structurally singular, the following equations cannot be solved for any variable:
	  0.0 = 6.28318530717959 * order5Type1_Inputs_Outputs.fn * (order5Type1_Inputs_Outputs.w + (-1.0))
	  pwLine1.n.ir + pwLine2.n.ir + pwLine3.p.ir + pwLine4.p.ir + pwLinewithOpening1.p.ir = 0.0
```

We can see that OpenModelica also [has some trouble](https://libraries.openmodelica.org/branches/master/OpenIPSL/files/OpenIPSL_OpenIPSL.Tests.Machines.PSAT.Order5test2.err), warning:

```
Warning: It was not possible to determine if the initialization problem is consistent, because of not evaluable parameters/start values during compile time: $DER.order5Type1_Inputs_Outputs.delta = 0.0 (order5Type1_Inputs_Outputs.w_b * (-1.0 + $START.order5Type1_Inputs_Outputs.w) = 0.0)
Warning: The initial conditions are over specified. The following 1 initial equations are redundant, so they are removed from the initialization system:
         $DER.order5Type1_Inputs_Outputs.delta = 0.0.
```

Unfortunately the messaging in System Modeler isn't very helpful to pinpoint a more specific place.

Maybe someone with knowledge of the model is able to improve the specification of the initialization to make it easier to solve?

gkhalley, 2024-02-10:
https://github.com/gkhalley/OpenIPSL/commit/d289e2ccdd543ac48106c77869144435e28667e0
It seems reasonable to comment out 'der(delta) = 0;' in the initial equation of Order5Type1.

dietmarw, 2024-02-12:
Thankse @gkhalley  I've now applied your fix (by removing it completely) in master.

commit a8c6758f (2024-02-12): Update Order5_Type1.mo

It seems reasonable to comment out "der(delta) = 0;" in the initial equation of Order5_Type1 to alleviate the over-specified condition.

Fixes #346
closed 2024-02-12 by dietmarw