OpenIPSL/OpenIPSL #360: PwFault yields SimCodegenUtil.superSystemVariability

gkhalley, 2024-11-27:
[18] 3:18 PM Validation of [PwFault](modelica:/OpenIPSL.Electrical.Events.PwFault) finished  
Internal Error: SimCodegenUtil.superSystemVariability: Failed to find any equation system variabilities. 
Validation of model OpenIPSL.Electrical.Events.PwFault completed with 1 error.

I don't understand this error in Wolfram System Modeler.

lvanfretti, 2024-11-28:
@maltelenz can you take a look?

maltelenz, 2024-11-28:
Thank you for this report!

I have created an issue internally on our side. Like all internal errors, it is clearly a bug in System Modeler.

It seems this may only be an issue for validation, and when the component is actually used in simulation models, the error doesn't occur?

Maybe this will allow you to ignore the issue for the moment.

lvanfretti, 2024-11-28:
@gkhalley and @maltelenz since we have identified this as a WSM bug and you have made an issue in your internal system, I will close down this issue and let you guys coordinate on resolving it since it is not really something that needs to be addressed at the library level.

closed 2024-11-28 by lvanfretti