OpenIPSL/OpenIPSL #364: A bug in the PSAT synchronous machine model OrderIV

jy369c14, 2025-02-15:
In the file "Order4.mo":
The initial value of e1d0 is not right. It can't guarantee that der(e1d)=0 at the initial point.
I suggest the lines 17-19 should be modified to:
`parameter Types.PerUnit vf00=V_MBtoSB*(e1q0 + (xd)*id0 + ra*iq0) "Initial value (system base)";
  parameter Types.PerUnit e1q0=vq0 + ra*iq0 + x1d*id0 "Initialization";
  parameter Types.PerUnit e1d0=(xq - x1q)*iq0 "Initialization";`
which can guarantee that the initial der(e1d)=0.

Moreover, the if-logic to deal with the case when xq=x1q doesn't work. The "if" and "else" should be swapped.
In the current code(lines 25-29), the following code means under whatever conditions, der(e1d) = (-e1d)/T1q0, which is obviously not right.
`if abs(xq-x1q) < Modelica.Constants.small then // safe-guard against division by zero.
    der(e1d) = ((-e1d) + (xq - x1q)*iq)/T1q0 "differential equations *";
  else
    der(e1d) = (-e1d)/T1q0 "differential equations *";
  end if;`
So lines 25-29 should be modified to:
`if abs(xq-x1q) >= Modelica.Constants.small then // safe-guard against division by zero.
    der(e1d) = ((-e1d) + (xq - x1q)*iq)/T1q0 "differential equations *";
  else
    der(e1d) = (-e1d)/T1q0 "differential equations *";
  end if;`

lvanfretti, 2025-02-27:
@marcelofcastro could you take a look?

jy369c14, 2025-02-28:
I apologize for the error in my previous suggestion. Upon further review, I realized that the correction I proposed was incorrect. The modified line 17 should be:

`parameter Types.PerUnit vf00=V_MBtoSB*(vq0 + ra*iq0+ (xd)*id0) "Initial value (system base)";`

lvanfretti, 2026-09-21:
[ut_machine_order4_mdl.zip](https://github.com/user-attachments/files/32462998/ut_machine_order4_mdl.zip)

- Bug investigated and addressed in PR #403. 
- Verification showed issue as described in PR #403, not the proposed fixes in this Issue.
- @jy369c14 see verification results in .zip attached.

closed 2026-09-21 by lvanfretti