OpenIPSL/OpenIPSL #367: Cyclical dependency in the parameters of `OpenIPSL.Tests.Machines.PSSE.GENTPJ`

qlambert-pro, 2025-03-27:
When validating the model `OpenIPSL.Tests.Machines.PSSE.GENTPJ`, Wolfram SystemModeler reports the following error:
> Error: OpenIPSL.Electrical.Machines.PSSE.GENTPJ [64:3-64:3] Invalid constant or parameter bindings. The values of Is, Xppdsat0, b, iq0, dsat0, a, delta0, DQ_dq, I_dq, id0, qsat0, Xppqsat0, Zs, PSIpp0, ang_PSIpp0, ang_PSIpp0andIt are cyclically dependent.
>  Is depends on: Zs, It, VT
>  Xppdsat0 depends on: dsat0, Xl, Xppd
>  b depends on: Xppdsat0, It.re, It.im, Xq
>  iq0 depends on: I_dq
>  dsat0 depends on: iq0, id0, S10, Kis, S12, PSIag
>  a depends on: dsat0, PSIag
>  delta0 depends on: b, a, ang_PSIpp0, ang_PSIpp0andIt
>  DQ_dq depends on: delta0
>  I_dq depends on: DQ_dq, It
>  id0 depends on: I_dq
>  qsat0 depends on: iq0, id0, Xd, S10, Kis, S12, PSIag, Xq
> Xppqsat0 depends on: qsat0, Xppq, Xl
>  Zs depends on: Xppqsat0, R_a
>  PSIpp0 depends on: id0, Xppdsat0, Is, Xppqsat0, Zs
>  ang_PSIpp0 depends on: PSIpp0
>  ang_PSIpp0andIt depends on: ang_PSIpp0, ang_It

I could confirm that there is a cyclical dependency between:
`delta0`, `DQ_dq`, `I_dq`, `id0`, `PSIpp0`, `and_PSIpp0`, `ang_PSIpp0andIt`.

This is invalid Modelica according to Section 4.4.4 of the Modelica specifications.

dietmarw, 2025-05-06:
@lvanfretti Is this something that one of your students can take a look at?

maltelenz, 2025-05-07:
I was also able to verify a cycle by hand, which I would be curious to see how other tools resolve.

I would dispute the `systemmodeler` label, since it seems to me that the other tools are the ones with the problem (not seeing the error) :) . I would be happy to be proven wrong though.

This is the cycle I verified:

`ang_PSIpp0andIt` -> `ang_PSIpp0`
https://github.com/OpenIPSL/OpenIPSL/blob/541d54c6f9d799fad3b69bfdee2246efe81a36e4/OpenIPSL/Electrical/Machines/PSSE/GENTPJ.mo#L69

`ang_PSIpp0` -> `PSIpp0.im`
https://github.com/OpenIPSL/OpenIPSL/blob/541d54c6f9d799fad3b69bfdee2246efe81a36e4/OpenIPSL/Electrical/Machines/PSSE/GENTPJ.mo#L71

`PSIpp0.im` -> `id0`
https://github.com/OpenIPSL/OpenIPSL/blob/541d54c6f9d799fad3b69bfdee2246efe81a36e4/OpenIPSL/Electrical/Machines/PSSE/GENTPJ.mo#L66

`id0` -> `I_dq.im`
https://github.com/OpenIPSL/OpenIPSL/blob/541d54c6f9d799fad3b69bfdee2246efe81a36e4/OpenIPSL/Electrical/Machines/PSSE/GENTPJ.mo#L103

`I_dq.im` -> `DQ_dq`
https://github.com/OpenIPSL/OpenIPSL/blob/541d54c6f9d799fad3b69bfdee2246efe81a36e4/OpenIPSL/Electrical/Machines/PSSE/GENTPJ.mo#L95

`DQ_dq` -> `DQ_dq.re`
`DQ_dq.re` -> `delta0`
https://github.com/OpenIPSL/OpenIPSL/blob/541d54c6f9d799fad3b69bfdee2246efe81a36e4/OpenIPSL/Electrical/Machines/PSSE/GENTPJ.mo#L93

`delta0` -> `ang_PSIpp0andIt`
https://github.com/OpenIPSL/OpenIPSL/blob/541d54c6f9d799fad3b69bfdee2246efe81a36e4/OpenIPSL/Electrical/Machines/PSSE/GENTPJ.mo#L90-L91

lvanfretti, 2025-05-07:
@maltelenz thanks for being so diligent! The hand check you did is very helpful. 
I agree that in this case is really not a WSL issue, but actually that of other tools. @gkhalley removed the `systemmodeler` tag, and I have removed the `toolspecific` tag as well.

I guess OM and Dymola are letting us get away with some things that the spec doesn't allow. For OM, you can test yourself by downloading it and testing the corresponding test model for this component. Meanwhile, for Dymola, I am not sure what information you are looking for, but let me know and I'll see if I can provide it.

I will try to test with Modelon Impact desktop too, usually they are also flagging cases that OM and Dymola don't.

In the mean time, I don't think I can resolve this, I'm going to be away for a few weeks so won't get back to this until later. I'll talk with @marcelofcastro to see if he can help.

maltelenz, 2025-05-07:
@lvanfretti At the moment I'm satisfied with the fact that it looks like System Modeler is doing the right thing. Only if it turns out that the model is somehow valid after all would I be interested in why and what the other tools are doing.

lvanfretti, 2025-05-07:
@maltelenz I would still like to change the model so this issue doesn't arise, as for example @gkhalley needs to use this model. I'll see what @marcelofcastro can do to help.

gkhalley, 2025-05-08:
Please see [https://github.com/gkhalley/OpenIPSL/commit/b1fa0d55f38053ed859521cd789b621a6ed2cbc6](url)
The build log implies there is still some work ... see below. I will continue to test. I'm interested in feedback ... like "this is completely wrong" or "this has possibilities."
Building "OpenIPSL.Tests.Machines.PSSE.GENTPJ"
Warning: Relation == on Real operands is not allowed outside functions: gENTPJ.S10 == 0.0
Warning: Relation <> on Real operands is not allowed outside functions: gENTPJ.S12 <> 0.0
Comes from function OpenIPSL.NonElectrical.Functions.SE

lvanfretti, 2025-05-08:
@gkhalley setting the start values to zero will lead to poor or failing initialization, for example,   setting ` delta(start = 0, fixed = true) ` instead of `  delta(start=delta0, fixed=true),` will set the internal angle of the machine to zero instead of the value from the power flow solution, which is what we need. All start values need to be pre-calculated as a parameter ` parameter Complex PSIpp0=Complex(real(Zs*Is),(imag(Zs*Is) - id0*(Xppqsat0-Xppdsat0))) ` is calculating the initial value of the flux based on the machine equations. 

I didn't analyze all other changes proposed by the AI... but as useful as it might be, we can't do what is suggestion other than for testing. Obviously the output gives you an idea that the parameters are the issue, but we can't just get rid of them like the AI is proposing.

What we need is to revise the initialization parameters so they are not creating a cyclical dependency, while at the same time providing a good starting guess for the initialization problem. Most likely this can be fixed by modifying how the parameters such as `PSIpp0` are being calculated by introducing auxiliary parameters or values. Before you go any further, the best exercise is to understand how we initialize the simplest model, e.g., the voltage source [here](https://github.com/OpenIPSL/OpenIPSL/blob/master/OpenIPSL/Electrical/Sources/SourceBehindImpedance/BaseClasses/baseVoltageSource.mo) where `delta0' and 'E0' are calculated with the approach described above, see lines 66 and 67 and how we arrive to them.

closed 2026-10-07 by dietmarw