OpenIPSL/OpenIPSL #368: start values are determined in the same block of equations as the variables themselves

maltelenz, 2025-03-27:
When trying to simulate some models in System Modeler, we get an error about start values that are solved in the same block of equations as the variable itself.

Here is an example from `OpenIPSL.Examples.DAEMode.SMIB_Examples.Example_1.Network3`:

```
Variables have `start` values that are determined in the same block of equations as the variables themselves: `G1.eXST1_1.TransducerDelay.y`
	Equations in block:

	B1.p.vr = $wsmCSE[16] * G1.gENSAL.uq + $wsmCSE[17] * G1.gENSAL.ud
	G1.gENSAL.uq = G1.gENSAL.PSId - G1.gENSAL.R_a * G1.gENSAL.iq
	G1.gENSAL.PSId = G1.gENSAL.PSIppd - G1.gENSAL.Xppd * G1.gENSAL.id
	$eventState3 := G1.gENSAL.EFD < (-3.0) * G1.eXST1_1.ECOMP0 - G1.gENSAL.XadIfd /* Event-generating expression */
	$eventState2 := 8.0 * G1.eXST1_1.ECOMP0 - G1.gENSAL.XadIfd < G1.gENSAL.EFD /* Event-generating expression */
	$eventState5 := G1.eXST1_1.feedback.y < G1.eXST1_1.limiter.uMin /* Event-generating expression */
	$eventState4 := G1.eXST1_1.limiter.uMax < G1.eXST1_1.feedback.y /* Event-generating expression */
	$eventState1 := G1.sTAB2A.transferFunction5.y < G1.sTAB2A.limiter.uMin /* Event-generating expression */
	$eventState0 := G1.sTAB2A.limiter.uMax < G1.sTAB2A.transferFunction5.y /* Event-generating expression */
	0.0 = ((if G1.sTAB2A.transferFunction3.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction3.a[1] ^ 2 + G1.sTAB2A.transferFunction3.a[2] ^ 2) then G1.sTAB2A.transferFunction3.a[2] else 1.0) * G1.sTAB2A.transferFunction2.y - G1.sTAB2A.transferFunction3.a[2] * G1.sTAB2A.transferFunction3.x_scaled[1]) / G1.sTAB2A.transferFunction3.a[1]
	G1.sTAB2A.transferFunction3.y = G1.sTAB2A.transferFunction3.b[1] * G1.sTAB2A.transferFunction3.x_scaled[1] / (if G1.sTAB2A.transferFunction3.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction3.a[1] ^ 2 + G1.sTAB2A.transferFunction3.a[2] ^ 2) then G1.sTAB2A.transferFunction3.a[2] else 1.0)
	0.0 = ((if G1.sTAB2A.transferFunction2.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction2.a[1] ^ 2 + G1.sTAB2A.transferFunction2.a[2] ^ 2) then G1.sTAB2A.transferFunction2.a[2] else 1.0) * G1.sTAB2A.transferFunction1.y - G1.sTAB2A.transferFunction2.a[2] * G1.sTAB2A.transferFunction2.x_scaled[1]) / G1.sTAB2A.transferFunction2.a[1]
	0.0 = ((if G1.sTAB2A.transferFunction1.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction1.a[1] ^ 2 + G1.sTAB2A.transferFunction1.a[2] ^ 2) then G1.sTAB2A.transferFunction1.a[2] else 1.0) * G1.sTAB2A.transferFunction.y - G1.sTAB2A.transferFunction1.a[2] * G1.sTAB2A.transferFunction1.x_scaled[1]) / G1.sTAB2A.transferFunction1.a[1]
	0.0 = ((if G1.sTAB2A.transferFunction.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction.a[1] ^ 2 + G1.sTAB2A.transferFunction.a[2] ^ 2) then G1.sTAB2A.transferFunction.a[2] else 1.0) * G1.gENSAL.PELEC - G1.sTAB2A.transferFunction.a[2] * G1.sTAB2A.transferFunction.x_scaled[1]) / G1.sTAB2A.transferFunction.a[1]
	pwFault.p.ir = if $eventState12 then 0.0 else if $eventState13 then 2220.0 * B2.p.vi else 0.0
	twoWindingTransformer.n.ir + pwLine1.p.ir + pwLine2.p.ir + pwFault.p.ir = 0.0
	if-equation no: 1, branch position -1
	if-equation no: 0, branch position -1
	pwFault.p.ii = if $eventState12 then 0.0 else if $eventState13 then (-2220.0) * B2.p.vr else 0.0
	twoWindingTransformer.n.ii + pwLine1.p.ii + pwLine2.p.ii + pwFault.p.ii = 0.0
	(twoWindingTransformer.Vn ^ 2 * twoWindingTransformer.rT * twoWindingTransformer.n.ii + twoWindingTransformer.Vn ^ 2 * twoWindingTransformer.xT * twoWindingTransformer.n.ir) * twoWindingTransformer.S_b / (twoWindingTransformer.Sn * twoWindingTransformer.V_b ^ 2) = B2.p.vi - B1.p.vi / twoWindingTransformer.m
	((-twoWindingTransformer.Vn ^ 2) * twoWindingTransformer.xT * G1.p.ir - twoWindingTransformer.Vn ^ 2 * twoWindingTransformer.rT * G1.p.ii) * twoWindingTransformer.S_b / (twoWindingTransformer.Sn * twoWindingTransformer.V_b ^ 2) = B1.p.vi * twoWindingTransformer.m ^ (-2) - B2.p.vi / twoWindingTransformer.m
	G1.p.ii = G1.gENSAL.M_b * ($wsmCSE[16] * G1.gENSAL.id - $wsmCSE[17] * G1.gENSAL.iq) / G1.gENSAL.S_b
	(twoWindingTransformer.Vn ^ 2 * twoWindingTransformer.xT * G1.p.ii - twoWindingTransformer.Vn ^ 2 * twoWindingTransformer.rT * G1.p.ir) * twoWindingTransformer.S_b / (twoWindingTransformer.Sn * twoWindingTransformer.V_b ^ 2) = B1.p.vr * twoWindingTransformer.m ^ (-2) - B2.p.vr / twoWindingTransformer.m
	(twoWindingTransformer.Vn ^ 2 * twoWindingTransformer.rT * twoWindingTransformer.n.ir - twoWindingTransformer.Vn ^ 2 * twoWindingTransformer.xT * twoWindingTransformer.n.ii) * twoWindingTransformer.S_b / (twoWindingTransformer.Sn * twoWindingTransformer.V_b ^ 2) = B2.p.vr - B1.p.vr / twoWindingTransformer.m
	-G1.gENSAL.P = G1.p.ii * B1.p.vi + G1.p.ir * B1.p.vr
	G1.gENSAL.PELEC = G1.gENSAL.P * G1.gENSAL.S_b / G1.gENSAL.M_b
	G1.sTAB2A.transferFunction.y = (G1.sTAB2A.transferFunction.b[2] - $wsmCSE[15]) * G1.sTAB2A.transferFunction.x_scaled[1] / (if G1.sTAB2A.transferFunction.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction.a[1] ^ 2 + G1.sTAB2A.transferFunction.a[2] ^ 2) then G1.sTAB2A.transferFunction.a[2] else 1.0) + G1.sTAB2A.transferFunction.b[1] * G1.gENSAL.PELEC / G1.sTAB2A.transferFunction.a[1]
	G1.sTAB2A.transferFunction1.y = (G1.sTAB2A.transferFunction1.b[2] - $wsmCSE[14]) * G1.sTAB2A.transferFunction1.x_scaled[1] / (if G1.sTAB2A.transferFunction1.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction1.a[1] ^ 2 + G1.sTAB2A.transferFunction1.a[2] ^ 2) then G1.sTAB2A.transferFunction1.a[2] else 1.0) + G1.sTAB2A.transferFunction1.b[1] * G1.sTAB2A.transferFunction.y / G1.sTAB2A.transferFunction1.a[1]
	G1.sTAB2A.transferFunction2.y = (G1.sTAB2A.transferFunction2.b[2] - $wsmCSE[13]) * G1.sTAB2A.transferFunction2.x_scaled[1] / (if G1.sTAB2A.transferFunction2.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction2.a[1] ^ 2 + G1.sTAB2A.transferFunction2.a[2] ^ 2) then G1.sTAB2A.transferFunction2.a[2] else 1.0) + G1.sTAB2A.transferFunction2.b[1] * G1.sTAB2A.transferFunction1.y / G1.sTAB2A.transferFunction2.a[1]
	G1.sTAB2A.K.y = G1.sTAB2A.K.k * G1.sTAB2A.transferFunction2.y
	G1.sTAB2A.add.y = G1.sTAB2A.add.k1 * G1.sTAB2A.transferFunction3.y + G1.sTAB2A.add.k2 * G1.sTAB2A.K.y
	0.0 = ((if G1.sTAB2A.transferFunction4.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction4.a[1] ^ 2 + G1.sTAB2A.transferFunction4.a[2] ^ 2) then G1.sTAB2A.transferFunction4.a[2] else 1.0) * G1.sTAB2A.add.y - G1.sTAB2A.transferFunction4.a[2] * G1.sTAB2A.transferFunction4.x_scaled[1]) / G1.sTAB2A.transferFunction4.a[1]
	G1.sTAB2A.transferFunction4.y = G1.sTAB2A.transferFunction4.b[1] * G1.sTAB2A.transferFunction4.x_scaled[1] / (if G1.sTAB2A.transferFunction4.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction4.a[1] ^ 2 + G1.sTAB2A.transferFunction4.a[2] ^ 2) then G1.sTAB2A.transferFunction4.a[2] else 1.0)
	0.0 = ((if G1.sTAB2A.transferFunction5.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction5.a[1] ^ 2 + G1.sTAB2A.transferFunction5.a[2] ^ 2) then G1.sTAB2A.transferFunction5.a[2] else 1.0) * G1.sTAB2A.transferFunction4.y - G1.sTAB2A.transferFunction5.a[2] * G1.sTAB2A.transferFunction5.x_scaled[1]) / G1.sTAB2A.transferFunction5.a[1]
	G1.sTAB2A.transferFunction5.y = G1.sTAB2A.transferFunction5.b[1] * G1.sTAB2A.transferFunction5.x_scaled[1] / (if G1.sTAB2A.transferFunction5.a[2] > 2.22044604925031e-14 * sqrt(G1.sTAB2A.transferFunction5.a[1] ^ 2 + G1.sTAB2A.transferFunction5.a[2] ^ 2) then G1.sTAB2A.transferFunction5.a[2] else 1.0)
	G1.sTAB2A.VOTHSG = homotopy(smooth(0, if $eventState0 then G1.sTAB2A.limiter.uMax else if $eventState1 then G1.sTAB2A.limiter.uMin else G1.sTAB2A.transferFunction5.y), G1.sTAB2A.transferFunction5.y)
	G1.eXST1_1.V_REF = $wsmCSE[12] + G1.eXST1_1.ECOMP0
	G1.eXST1_1.VoltageReference.k = G1.eXST1_1.V_REF
	guess(G1.eXST1_1.TransducerDelay.y) := G1.eXST1_1.TransducerDelay.y_start /* Parameter equation */
	G1.eXST1_1.TransducerDelay.y_start = G1.eXST1_1.ECOMP0
	G1.eXST1_1.TransducerDelay.y = if abs(G1.eXST1_1.TransducerDelay.T) <= 2.2204460492503131e-16 then G1.eXST1_1.TransducerDelay.K * G1.eXST1_1.ECOMP0 else G1.eXST1_1.TransducerDelay.y_start
	G1.eXST1_1.DiffV.y = G1.eXST1_1.DiffV.k1 * G1.eXST1_1.VoltageReference.k + G1.eXST1_1.DiffV.k2 * G1.eXST1_1.TransducerDelay.y
	G1.eXST1_1.feedback.u1 = G1.eXST1_1.add3_2.k1 * G1.sTAB2A.VOTHSG + G1.eXST1_1.add3_2.k2 * G1.eXST1_1.DiffV.y + G1.eXST1_1.add3_2.k3 * G1.eXST1_1.Limiters.y
	G1.eXST1_1.feedback.y = G1.eXST1_1.feedback.u1 - G1.eXST1_1.imDerivativeLag.y_start
[... cut ...]
	G1.p.ir = ((-$wsmCSE[16]) * G1.gENSAL.iq - $wsmCSE[17] * G1.gENSAL.id) * G1.gENSAL.M_b / G1.gENSAL.S_b
	0.0 = ((G1.gENSAL.Xq - G1.gENSAL.Xppq) * G1.gENSAL.iq - G1.gENSAL.PSIppq) / G1.gENSAL.Tppq0
	G1.gENSAL.PSIq = -G1.gENSAL.PSIppq - G1.gENSAL.Xppq * G1.gENSAL.iq
	G1.gENSAL.ud = -G1.gENSAL.PSIq - G1.gENSAL.R_a * G1.gENSAL.id
	B1.p.vi = $wsmCSE[17] * G1.gENSAL.uq - $wsmCSE[16] * G1.gENSAL.ud
	G1.eXST1_1.ECOMP0 = sqrt(B1.p.vi ^ 2 + B1.p.vr ^ 2)
```

Do you have any comments from the library side?

lvanfretti, 2025-03-27:
Hi @maltelenz 
Not sure what comments you are looking for...
The model doesn't give any similar complain in Dymola (even using pedantic check) and simulates fine:
[image]

maltelenz, 2025-03-31:
I guess I am looking for information like:

* Have you had any issues with solving the initialization problem for this model before or in other tools?
* Is this a property you would expect from your model (having to solve the start value in the same block of equations as the variable itself)?
* Do you see any changes that could be made in the model to make the initialization problem easier to solve?

lvanfretti, 2025-03-31:
Hi @maltelenz 

- Neither OM or Dymola seem to have any issues. Impact is able to initialize but halts when the fault block is active: 

[image]
 
[image]

- It is a bit difficult to answer without knowing what specific start value is of concern. Can you be more specific?

- Well, I don't think is really a problem of the model... I guess if the problem is with the start values, we could ask for them to be fixed=false? But again, without anything specific to target, I wouldn't be able to advice.

maltelenz, 2025-04-02:
Thanks @lvanfretti for your comments.

There are 19 models in OpenIPSL that fail with similar messages in System Modeler.

For this specific one, I was able to simplify (or break apart I guess) the initial problem sufficiently by editing this line:

https://github.com/OpenIPSL/OpenIPSL/blob/9aa77e0a50e5209f7349fdc896be81512de1dcb1/OpenIPSL/NonElectrical/Continuous/SimpleLag.mo#L8

and adding an `annotation(Evaluate = true)`.

That made the system build and simulate in System Modeler.

For some other models, I added the same annotation to other `Time` parameters in neighboring classes which made them simulate as well. There are however still some models where I haven't found such a change that works (at least yet).

Does such a change seem reasonable to you? If more parameters are evaluated, the equation systems can be simplified more and become easier (and potentially faster) to solve, at the cost of less flexibility in changing these parameters later at the simulation stage.

I noticed some cases of `Evaluate = false` in the `SimpleLead` and `SimpleLagRateLimBlock` classes, so it seems these already have been tweaked for some purpose in the past.

lvanfretti, 2025-04-02:
@maltelenz  I think it would be beneficial if you share with us your list and results of the problems.
- In principle I don't think is a major issue to add these kind of annotations if there is a benefit, however, there is the trade off then that the model would not allow for parametric studies that easily or without removing the annotation.

- I would like to bring in @dietmarw to this discussion to see what is his take.

- One thought is that such annotation should perhaps be possible to enable/disable by user selection, perhaps via a replaceable when the component is used, so that the user is aware that there is a trade off when using the model in such way.

maltelenz, 2025-04-03:
I pushed a branch with changes here: https://github.com/OpenIPSL/OpenIPSL/compare/master...maltelenz:OpenIPSL:bugfix/parameter-evaluation

These two changes "fix" these models in System Modeler, to where they simulate:

```
OpenIPSL.Tests.Controls.PSSE.ES.ST5B
OpenIPSL.Tests.Controls.PSSE.ES.ESST4B
OpenIPSL.Tests.Controls.PSSE.ES.EXST1
OpenIPSL.Examples.RaPIdExperiments.Mostar
OpenIPSL.Examples.DAEMode.SMIB_Examples.Example_1.Network3
OpenIPSL.Examples.SevenBus.Network
```

However, we still have the kind of message reported in the starting comment of this issue for a larger selection of models. For these, I haven't been able to find similar changes that "fix" them:

```
OpenIPSL.Examples.Tutorial.Example_4.Experiments.SMIB
OpenIPSL.Examples.Tutorial.Example_4.Experiments.SMIBVarLoad
OpenIPSL.Tests.Controls.PSSE.ES.AC7B
OpenIPSL.Tests.Controls.PSSE.ES.AC8B
OpenIPSL.Tests.Controls.PSSE.ES.ESAC2A
OpenIPSL.Tests.Controls.PSSE.ES.ESST1A
OpenIPSL.Tests.Controls.PSSE.ES.EXNI
OpenIPSL.Tests.Controls.PSSE.ES.SCRX
OpenIPSL.Tests.Controls.PSSE.ES.URST5T
OpenIPSL.Tests.Controls.PSSE.PSS.IEEEST
OpenIPSL.Tests.Controls.PSSE.PSS.PSS2B
OpenIPSL.Tests.Controls.PSSE.TG.HYGOV
OpenIPSL.Tests.Controls.PSSE.TG.IEESGO
```

dietmarw, 2025-04-03:
@maltelenz I would not worry about any of the `*.Tests.*` models since they really are just development tests of very differnt majurity. So not to be mistaken as real designed test like `ModelicaTest`. We basically just kept them for keeps sake. `*.Examples.*`  are the ones that really should work and we should care about.

dietmarw, 2025-04-03:
@maltelenz Feel free to create a PR based on your branch.

dietmarw, 2025-04-07:
@maltelenz  With that PR merged, would it be OK to close this discussion, or do you still see something that needs to be discussed further?

maltelenz, 2025-04-08:
System Modeler still has trouble with the two `SMIB` examples mentioned.

It seems from the library developer side, you would expect the current models to work (and they do seem to work in two other tools), so my takeaway is that this is something you would consider a tool issue unless we can provide evidence to the contrary :)

Based on that, I would say that yes, this discussion can be closed.

closed 2025-04-08 by maltelenz