OpenIPSL/OpenIPSL #372: issue with RampTrackingFilter

johhell, 2025-09-17:
`OpenIPSL.NonElectrical.Continuous.RampTrackingFilter`

I tried to run an small `PSS2B` test and noticed a discrepancy in the results due a problem in the `RampTrackingFilter` block.

### original code
In the mentioned block 2 transfer functions are used:

```modelica
Continuous.TransferFunction TF1[M](b=fill({1},M), a=fill({T_2,1},M),each y_start=y_start) ....
Continuous.TransferFunction TF2[N](b=fill({T_1,1},N), a=fill({T_2,1},N), each y_start=y_start) ...
```
connection of blocks for N=1
```modelica
  elseif N == 1 then
    connect(u, TF2[1].u);
    connect(TF2[1].y, TF1[1].u);
    for i in 1:M-2 loop
      connect(TF1[i].y, TF1[i+1].u);
    end for;
    connect(TF1[M-1].y, TF1[M].u);
    connect(TF1[M].y, y);
```
### test
in my example I used 
```modelica
  parameter Real T_8=0.5;
  parameter Real T_9=0.1;
  parameter Integer M = 5;
  parameter Integer N = 1;
```
With `M=5`  5x TF1-blocks + 1x TF2 block are connected in series. But in the denominator of TF2 the time constant `T_2` is included as well. 
That means we have in summary 1x numerator with `T_1` and 6x denominator with `T_2`. And this is not correct.

Below a comparison with transfer function from `ControlSystemsBase` tested with Julia. 

```julia
## ramp tracking
rt1 =  (tf([1],[T9,1])^M * tf([T8,1],[1]))^N
y1, t1  = lsim(rt1, eIn, tt)
plot!(plt1,t1,y1[1,:], label="TF-lsim M=5")

```

[image]

all files here: [testRT.zip](https://github.com/user-attachments/files/22389901/testRT.zip)

dietmarw, 2025-10-16:
The question for me is, is that behaviour simply not correct or just not what you expected. Unfortunately the documentation is not helping here, @GiuseppeLaera @lvanfretti can you put som light on the **intended** implementation?

dietmarw, 2025-10-17:
I suspected as much. Maybe we can include the explaination you have just given here in a shortened version in the infolayer.

lvanfretti, 2025-10-17:
@dietmarw  good idea, I've listed in #371

AHaumer, 2025-10-18:
@johhell a correct Modelica implementation you'll find in the enclosed zip, using LinearSystem2 for correct calculation of numerator and denominator polynomial of the transfer function.

[Test.zip](https://github.com/user-attachments/files/22986682/Test.zip)

AHaumer, 2025-10-18:
@johhell I can even offer an implementation compliant with IEEE421.5 avoiding the usage of LinearSystems2 (see enclosed test).

[Test.zip](https://github.com/user-attachments/files/22986905/Test.zip)

AHaumer, 2025-10-19:
@lvanfretti I have little expertise in the field covered by OpenIPSL, but I know a little bit about transfer functions.

Do you agree that IEEE 421.5 as well as Figure 5, 6 and first line of Figure 7 of the already mentioned publication show:

[image]

whereas the current implementation in OpenIPSL and the rest of Figure 7 of the already mentioned publication show: 

[image]

The two transfer functions are mathematically not identical, and they deliver different results.
I admit that for special cases (like M=0 and / or N=0) there is the need of interpretation (hat is desired), but for normal cases not.

@dietmarw what's your opinion about the transfer functions (although for me it is mathematically clear)?

AHaumer, 2025-10-19:
Here is my proposal, you may transfer the block to OpenIPSL if you would like to do so.

[TestRampTrackingFilter.zip](https://github.com/user-attachments/files/22991451/TestRampTrackingFilter.zip)

Here is a comparison for M=5, N=1, T_1=0.5, T_2=0.1

[image]

AHaumer, 2025-10-19:
BTW compare the result of my Test-example with Figure 9 of the already mentioned publication:

[image]

dietmarw, 2025-10-19:
@AHaumer I don't really have much to add to what @lvanfretti  already has tried to point out multiple times. The RampTracking filter is not supposed to be the one from the IEEE standard but tries to represent the same behaviour the PSS/E implementations shows. This is what @lvanfretti and his PhD students have been trying to accomplish. 
If you think that having the implementation as exactly specified in the standard has some practical value (remember, people might not actually want that behaviour but simply just the same result as PSS/E) then we are more than happy to accept a PR adding the IEEE standard version in **addition** to the currently present version.

johhell, 2025-10-20:
Before closing this issue I would like to add some infomation

### in general:
* I have not read all of your comments from last weekend.
* I am 99.99% sure that the PSS implementation PSS/E is correct, that means according IEEE421.5

### the good news:
* I will try to provide models for PSS2A, PSS2B, and PSS2C, including test cases in an additional folder.
* time frame: within 2025
* Before I do this, I will try to run the corresponding test cases `/OpenIPSL/Tests/Controls/PSSE/PSS/` for `PSS2A` and `PSS2B` in my OM installation   [`OpenModelica Compiler v1.26.0-dev-265-g885e18d0b9-cmake`]. 

BUT
* test PSS2A failed during initialization
* test PSS2B was running, but here parameters (M=0, N=0 ??). When changing to M=5, N=1 (which is realistic) then simulation failed as well.

### Background
I am currently trying to find a solution to the initialization problem in power system studies.
My approach here (README only):  https://github.com/johhell/EPSdynamix.jl

The idea behind: A block-by-block initialization in a monolithic model - similar to FMU's. And it works. Written in JULIA, based on MODIA. To test my tool, I used the results of various tools, such as OpenIPSL.
Unfortunately, Hilding and Martin have stopped maintaining MODIA and are now working on new projects (bad luck for me, but the idea of having some kind of subsystem like this in MODELICA would be great).

The key advantage I see:
* one language for simulation, pre- and post-processing.

Johann

AHaumer, 2025-10-22:
@lvanfretti I understand that you aim for a good match with PSS/E-results. 
But do you agree that Figure 7 of the mentioned paper is not correct?

[image]

Swapping the two transfer functions can't influence the result. 
And the total transfer function of the series connection is not as shown above but:

[image]

Therefore it's a wise decision to change the misleading icon of the block.